timetag/ETA #13: Notify user if essentials are missing in the Instument designer

Cyberfly100, 2018-07-25:
like when an initial state is not defined there should be a warning. So far it is only showing an error on compile time. I think a full compilation is not necessary, just a check, that each arrow is labled and that an initial arrow exists.

linzuzeng, 2018-07-25:
We could also do a full check by running a partial-compilation.  We can jump the code generation part and only performs the graph checking and grammar checking for ETA code.

linzuzeng, 2019-10-19:
I think ETA shows warning when clicking the file browsing button.
But I also find it doesn't show user-understandable error messages. For example, when initial is not set, it shows a long error message on the backend only.

We need to add more checks to the backend when dealing with the user inputs.

linzuzeng, 2020-02-23:
Fixed.

closed 2020-02-23 by linzuzeng