timetag/ETA #15: automatically create header/comment for items added via the "create" menu

Cyberfly100, 2018-07-25:
The resoning is that users that use the create menu are not super familliar with the program, so it should provide some more info.

E.g.:
when a histogram with name h1 and dimension (10000,16) is created it should show up as

HISTOGRAM(h1,(10000,16)) #HISTOGRAM(name, (number of bins, bin size))

this could also be done as a tooltip, but i assume that that would be way more difficult

linzuzeng, 2018-07-25:
The create button should really be improved. 

> HISTOGRAM(h1,(10000,16)) #HISTOGRAM(name, (number of bins, bin size))
This is easy to fix. Maybe we can also  have a live-preview of the code to be inserted.

> This could be extended with a small icon (interactive?) to show what the function does 
That is easy, we can even have animations on the CREATE dialog. But we need some nice document there. Could you help me with this part?

linzuzeng, 2018-08-01:
fixed in RC5.2

closed 2018-08-01 by linzuzeng