timetag/ETA #19: display variables in code panel and instrument designer

Cyberfly100, 2018-07-25:
i think they would like this like in matlab / spyder. But it could of course also look pretty :)

linzuzeng, 2018-07-25:
This must be a lovely feature, but also hard.

At the ETA Back-end, thanks to the check-pointing support since ETA RC4.x, we can now access all the contexts. Therefore this feature is possible to implement.
At the GUI side, it is easy to display the new thingy, INTEGER(). But it should be very ugly if we dump the array as string.

Also, do you also want it to be updated in real-time when the analysis is running?

Cyberfly100, 2018-07-25:
what I meant is the variables from the home screen. variables as in not instument designer, code panel or hardware. Like binsize and the filename...

linzuzeng, 2018-07-26:
Aha. Then it is easy.

Maybe we should call it "parameter" instead of "variable" to reduce the confusion.

closed 2018-07-27 by linzuzeng