timetag/ETA #20: make record smart

Cyberfly100, 2018-07-25:
could record distinguish by itself whether a multi-clock is used to automatically function as record_all?

linzuzeng, 2018-07-25:
Easy to fix.

Function overloading was supported since pre-RC era, but I didn't do it for record_all because I thought there might be different kinds of record_???, ( say, record_last, record_first ).
However, later we invent find_start() for Gunnar clock, so we don't need record_??? anymore. 
Sure we can now safely overload record to record_all.

The concept of multi-clock might be useless in ETA-M.  However, we can still keep multi-clock so as to ease users from craziness of multiverse.

linzuzeng, 2018-08-06:
fixed in RC5.3

closed 2018-08-06 by linzuzeng