timetag/ETA #21: drop the group feature

Cyberfly100, 2018-07-25:
I said we never used it and the multiple recipe scheme kind of makes is superfluous. 
Thay say unnesessary features should be dropped before a small percentage of the user base starts to like it and gets angy when they are finally dropped because of incompatibility.

linzuzeng, 2018-07-25:
Easy to fix.

However, are you really sure that you don't want it?
I thought it could be used for performing two different kinds of analysis on a same file. For example, we can do the correlation while seeing the stability (power-meter mode).

Cyberfly100, 2018-07-25:
yes I know. I also kind of like it, but I end up just loading the correlation recipe and afterwards loading the power-meter recipe. It is annoying to type the filename again, but it makes the recipes less crowded

linzuzeng, 2018-07-26:
OK, then we drop it.

Cyberfly100, 2018-07-28:
hide by default, with option to enable in settings

linzuzeng, 2019-02-12:
found a reason for using it. not dropping

closed 2019-02-12 by linzuzeng