timetag/ETA #22: add a browse functionality for adding a file-variable

Cyberfly100, 2018-07-25:
the file variable would then obviously be a special variable since no one wants to browse for the binsize, but a file is needed for every analysis anyways. Maybe it could already be there in an empty recipe.
For batch mode we can just split the folder off of the file, then the user has to select a random file in the folder that he/she wants the batch mode to be run on.

linzuzeng, 2018-07-26:
Agree. I am already annoyed by copy and pasting the file name.  

Could you maybe send me your python code in Tkinter that opens the dialog for file/folder selection? I can add it in the next version.

Also note that file selection will work nicely only when the back-end and front-end are at the same computer. Otherwise user must RDP into the remote back-end to see the GUI. We are not going to make our own RDP.

Cyberfly100, 2018-07-28:
import tkinter as tk
from tkinter.filedialog import askopenfilename

root = tk.Tk()
root.update()
path = askopenfilename(filetypes=[("map files","*.2dmap")],initialdir='K:/')
root.destroy()

linzuzeng, 2018-08-01:
fixed in RC5.2

closed 2018-08-01 by linzuzeng