timetag/ETA #33: List the necessary packages in a conda environment

gyger, 2018-08-10:
It would be nice if the necessary packages to use my "private" site-packages folder are listed in the documentation.

gyger, 2018-08-10:
I used the following commands to get a running environment in conda.

- `conda create -n ETA python numpy numba cython llvmlite cffi flask decorator nbformat pytz requests retrying`
- Set the side package folder in ETA_LIB path
- `C:\Users\gyger\Anaconda3\envs\ETA\Lib\site-packages`
- `pip install astunparse`
  - Unfortunately the current version uses the variable name async, not allowed in Python 3.7 was fixed in a commit (https://github.com/simonpercivall/astunparse/commit/72bcac2fc408408c7cd07043c392fd9932714604) but not released yet.
   - So one needs to fix this file manually currently.
- `pip install dash dash-renderer dash-html-components dash-core-components`

Afterwards I can run the program in the command line. I still get a funny error using the packaged software about numpy not being able to import multiarray.

linzuzeng, 2018-08-17:
I think that the packaged ETABackend.exe can not find the DLLs for conda-modified version of numpy.
It seems that conda changed the DLL loading behavior of numpy, and simply setting a sys.path is not enoguh for that.

linzuzeng, 2018-08-17:
I updated the documentation for dependencies of ETA.

https://eta.readthedocs.io/en/latest/python.html#using-your-own-python-site-packages

linzuzeng, 2019-04-30:
ETA can be installed via pip now.

```
pip install etabackend
python -m etabackend
```

linzuzeng, 2019-06-12:
Now ETA dropped the stupid packaged "ETABackend.exe" completely. 
Instead, it uses `pip install etabackend` like other packages in Python world. 😄

gyger, 2019-06-12:
That is awesome to hear. Very nice, Zuzeng.

gyger, 2019-06-30:
Good, I installed it today using a conda Environment and pip and everything worked as wanted.

closed 2019-06-30 by gyger