timetag/ETA #37: The simple_lifetime recipe is very broken

Cyberfly100, 2018-08-12:
plotly is not working because it is trying to plot a 2d hist (has been updated on github, but seemingly has not made it into the new version yet.
bokeh is anyways broken and should be replaced by matplotlib
in the instrument designer the X lifetime uses find_start(SYNC) instead of the new start(SYNCS) or start(LAST_SYNC) or infer_start_from_stop().

linzuzeng, 2018-08-17:
1. I triggered a new build, the plotly code should be fixed now.
2. Shall we drop support of bokeh?
3. It should be infer_start_from_stop(SYNC).

Cyberfly100, 2018-09-12:
1) plotly seems to work again
2) bokeh in simple_lifetime has been replaced with matplotlib
3) works.

done.

closed 2018-09-12 by Cyberfly100