timetag/ETA #39: TimeTagging file of SwabianInstruments does not work

gyger, 2018-08-13:
The time tagging file of Swabian Instruments does not work. The Software crashes.
I attached a file that was created using .dump() of the Swabian Instruments guys, with a signal on channel 1 and channel 2 and with the clock on Channel 7. The clock is divided by factor 8, just for you to know.

[20180813-QD_Lifetime.zip](https://github.com/timetag/ETA/files/2284406/20180813-QD_Lifetime.zip)

linzuzeng, 2018-08-17:
I tested it with the built-in recipe `correlation` and it seems to be working.
Please note that for SwabianInstruments we need a special argument `format=1` to distinguish it from quTAG.

Try changing the line to:
`cut=eta.simple_cut(file,format=1)#or format=FORMAT_SI for the latest version`

linzuzeng, 2018-08-17:
Found the bug.  It was a typo which causes negative pointer in timetagfile.

https://github.com/timetag/ETA/commit/e572c186a6513842823638590c7f85f0bb991d27

linzuzeng, 2018-08-17:
Fixed in RC 5.4.

closed 2018-08-17 by linzuzeng