timetag/ETA #41: Browse window opens without focus

Cyberfly100, 2018-08-15:
If browse window is clicked for the second time after program start (scenario: measure a series of something - analysis stays the same but file chnages) the open file dialogue is opened behind the frontend.

linzuzeng, 2018-08-17:
Actually  it is opened with focus, and it works fine on Windows 10.  It could be a problem of tkinter.

```
            import tkinter as tk
            from tkinter.filedialog import askopenfilename
            root = tk.Tk()
            root.update()
            root.withdraw()
            root.attributes("-toolwindow", 1)
            root.wm_attributes("-topmost", 1) # open window with focus
            path = askopenfilename(filetypes=[("Time Tag File", "*.*")])
            root.destroy()
```

linzuzeng, 2018-09-12:
fixed with 

```
root = tk.Tk()
root.update()
root.withdraw()
root.overrideredirect(True)# Make it impossible to close the root via clicking |X| or alt+F4.
root.geometry('0x0+0+0')# Make it almost invisible - no decorations, 0 size, top left corner.
root.attributes("-alpha", 0.3)# Make it more invisible
root.attributes("-toolwindow", 1)
root.wm_attributes("-topmost", 1)
root.deiconify() #show root again
root.lift() #make sure it is on top
root.focus_force() #and has focus
path = askopenfilename(parent=root, filetypes=[("Time Tag File", "*.*")])# parent=root makes sure the dialogue is inherits the roots attributes, like being on-top
root.destroy()

```

closed 2018-09-12 by linzuzeng