timetag/ETA #88: Integrate into jupyter lab

gyger, 2019-07-01:
This is just an idea, to make ETA usable inside jupyter lab and continue working with the results in the notebook below.

linzuzeng, 2019-07-10:
Very good idea! The Python part of ETA should be able to work without too many modifications. I will write some document for how to call those ETA APIs like eta.run() from any Python environment without using the the ETA Backend.

But I am worrying that the existing GUI might not work easily as a [Jupyter Front-End Shell](https://jupyterlab.readthedocs.io/en/stable/developer/extension_dev.html).

I would recommend to use [ETA Web GUI](https://timetag.github.io/ETA/gui/src/renderer/) for now.

gyger, 2022-03-19:
I just found [JDaviz](https://github.com/spacetelescope/jdaviz), that is a spectra analysis platform within jupyter. Could be useful if we want to look at how they implement gui parts that work within jupyter.

linzuzeng, 2022-03-20:
Sounds like a lot of work. Doesn't bokeh work in jupyter?

gyger, 2022-03-21:
No that works fine, just in case we want to have the state machine designer within jupyter. I think for now we can leave that issue closed. Just wanted to have that example documented.

closed 2022-03-21 by gyger