timetag/ETA #89: eta_runtime reads the wrong item in the cut definition for ByteSize

gyger, 2019-09-09:
I don't know when it happened, but it seems the incremental_cut function is broken.
The reason is in the read out of Byte per entry. 
This was set to be cut[0][-3], this is not the correct entry anymore. It should be either
`cut[0][-4]` or `cut[0][5]`.
An example of such a cut definition is given below:
`[[40, 40, 1, 1, 1249, 10, 0, 0, 'C:/FilePath']]`

To avoid an error like this in the future, perhaps the indexes could be stored as constants?

linzuzeng, 2019-09-10:
Thank you for finding the cause of the bug!
This array constructed so that its first 8 items (0 to 7) could be easily read from C side using a struct.
https://github.com/timetag/ETA/blob/master/etabackend/cpp/PARSE_TimeTags.cpp#L55

linzuzeng, 2019-09-10:
I further looked into the version history. The bug was intruduced in v0.6.2, and partially fixed in 
https://github.com/timetag/ETA/commit/cf51282a32c5c4db4daa152405499a8fd12b1311#diff-939fee4682bb0eaa7d33cacc4f7a1c90R243
But I forgot to do the same thing to incremental_cut().  Now I store them as constants, which should prevent this from happening.

Also I noticed that 
```
long long TTRes_pspr ;//2
long long DTRes_pspr ;//3
long long SYNCRate_pspr ;//4 
long long BytesofRecords ;//5
long long RecordType ;//6
```
don't really has to be in this struct because they are never changed during analysis (unless someone really want to glue two timetags taken with different instruments together). They will be removed and replaced with constants.

Cyberfly100, 2019-10-18:
tested, it works.

closed 2019-10-18 by Cyberfly100