timetag/ETA #92: Add pytest and CI e.g. Travis or whatever we like

gyger, 2019-09-10:
I think it is time to add pytest testcases for a lot of the code.

linzuzeng, 2019-09-11:
Yes. ETA has been using [appveyor](https://ci.appveyor.com/project/linzuzeng/etabackend) for a while for building binaries for Windows.  It also supports testing, I think.
But we can also use add Travis for building and testing on Linux. 

We have a [sample file](https://github.com/timetag/timetag.github.io/releases/tag/samplefile). We can run all the recipes with this sample file and compare the results. This will tell us if the commits break any built-in recipe. 

Later we can add testcases that covers each Action and Tools, etc. But I will take a while to cover the entire ETA, including frontend and backend.

gyger, 2019-09-11:
Appveyor is also good.

linzuzeng, 2019-10-19:
[Lifetime with delay line error chekcers.zip](https://github.com/timetag/ETA/files/3746221/Lifetime.with.delay.line.error.chekcers.zip)

I made the first test for the dealy line, since it was wronged for many times.

linzuzeng, 2019-12-20:
ETA now supports Linux! We can add test scripts to it later.

gyger, 2020-01-31:
@linzuzeng Is it possible to run a recipe just from the backend?

linzuzeng, 2020-01-31:
I guess you want to run it without backend (websocket server).
try:
``
from etabackend.eta import ETA
eta=ETA()
eta.compile_eta(etaobj)
```
etaobj is the parsed JSON file.
Then you can call ``eta.run()`` like in a script panel.

gyger, 2020-02-04:
> I made the first test for the delay line, since it was wronged for many times. (see #57)
I tried to run this recipe with the test T3 file and I get only errors, so I am not sure how this test delay line should work.

linzuzeng, 2020-02-05:
> 
> 
> > I made the first test for the delay line, since it was wronged for many times. (see #57)
> > I tried to run this recipe with the test T3 file and I get only errors, so I am not sure how this test delay line should work.

The ``DelayLineChecker`` works by inspecting the events that goes into and out of the delay line. It checks if the timetag after sort are monotonically increasing, and if the events on chn1, which goes into the delay line, will come out from chn3, at the correct time. 

Since it only remember 1 event, it may not find all possible types errors of the delay line. A better test can be conducted after implementing WFILE, where we can print the sorted timetag and compare it with the correct one.

The ``graph_comparator`` is for comparing two histograms in txt files, which is unrelated to this topic.
The ``txt+graph`` will run the checker with a full timetag file (with chn 0 and chn1). It shows no error in the version 8b4062c2c0d709b2b3630d30f827b0c41af375a8
The ``pyplot`` will run the checker with the file cut into four Clips. There is a bug here which reports some false error. The virtual instrument ``DelayLineChecker`` is not correctly written to include its own variables in the context. The variables in embedded Python will lost at the boundary in each Clips. Therefore, three false errors are shown as expected. 

Changing

```
{{{
    print("SYNCRate_pspr",SYNCRate_pspr)
    last_tag=0
    last_event_goes_into_delayline=0
}}}

```
to 
```
INTEGER(last_tag)
INTEGER(last_event_goes_into_delayline)
{{{
    print("SYNCRate_pspr",SYNCRate_pspr)
}}}
```
will make ``DelayLineChecker``  correct, and no errors will be shown. See attachment below for the updated recipe.
[Lifetime with delay line error chekcers.zip](https://github.com/timetag/ETA/files/4157321/Lifetime.with.delay.line.error.chekcers.zip)

gyger, 2020-02-05:
Hej,
I downloaded the new file and run it in the current HEAD and it spills a lot of errors on the terminal. Can you check with the file in ´tests/hydraharp/data/200204_hydraharp_T3_g2_320MHz.timeres´ if this also gives errors for you.

linzuzeng, 2020-02-05:
I was using a different file. I tried your file and also saw the errors. I am still looking into this.

linzuzeng, 2020-02-05:
I checked the file that you used. It seems that the file is corrupted, because many events are not following monotonically increasing order. I modified the checker to only check events on CHN0 and CHN1 from file. 
[File error chekcers.zip](https://github.com/timetag/ETA/files/4158904/File.error.chekcers.zip)
 
The checker shows no error on this [sample file](https://github.com/timetag/timetag.github.io/releases/download/samplefile/HH400_T3_80MHz_HBT.ptu).

linzuzeng, 2020-02-05:
Maybe we can start with the [sample file](https://github.com/timetag/timetag.github.io/releases/download/samplefile/HH400_T3_80MHz_HBT.ptu) for testing the [built-in recipes](https://github.com/timetag/ETA/blob/master/gui/src/renderer/js/recipes/) with hydraharp T3. 

I ran the ``create txt (HHT3)`` task on my side, please find attached for the analysis results.
[HH400_T3_80MHz_HBT_correlation.zip](https://github.com/timetag/ETA/files/4158954/HH400_T3_80MHz_HBT_correlation.zip)

gyger, 2020-02-05:
@linzuzeng I already added some testcases for the default correlation recipe but I guess I will move it to this file 80MHz for the T3 mode.

linzuzeng, 2020-02-05:
Sorry, I forgot to check. Now I see them. :)

gyger, 2020-02-05:
The sample file is quite large, so I am not sure if we want to put it into the git. Perhaps we can make a smaller one in the future for the 80MHz laser. I was thinking on the order of 2MB.

gyger, 2020-02-05:
@linzuzeng Can you integrate the tox test into the appveyor file. I am not fully seeing through the configuration. The last commit should be the tox file, so somehow with https://www.appveyor.com/docs/lang/python/ this should be configurable.

linzuzeng, 2020-02-06:
The testing [works nicely](https://ci.appveyor.com/project/linzuzeng/etabackend). Great!

gyger, 2020-02-06:
I added the requirements file to the tox-file environment and now it seems to crash. we can remove this again, but not sure if it is really the error.

linzuzeng, 2020-02-07:
> I added the requirements file to the tox-file environment and now it seems to crash. we can remove this again, but not sure if it is really the error.

I tried to remove it and the testing works again. It seems that tox cannot find this requirement.txt.

linzuzeng, 2020-03-11:
test works very well :) 
close for now

closed 2020-03-11 by linzuzeng