timetag/ETA #94: critical errors can get the frontend stuck, now

Cyberfly100, 2019-09-17:
A strange file caused the backend to have overflow errors, which caused the frontend to be stuck in the "working" animation. 
[image]

linzuzeng, 2019-09-19:
@Cyberfly100  Could you help me testing if it is fixed or not? You can try manually closing the back-end during an analysis.

Cyberfly100, 2019-10-15:
If I close the backend during analysis the frontend is still in the "ETA Running..." state and does not recover.
[image]

linzuzeng, 2019-10-15:
Strange that the [Web GUI ](https://timetag.github.io/ETA/gui/src/renderer/) shows an error message and the close button when Backend crashed. Could be a Electron bug. I will try to update Electron the latest version.

linzuzeng, 2019-10-15:
Confirmed. The error message shows up after updating the Electron from 4.0 to 6.0. 
However, the message shows up multiple times, since ETA Frontend has a tiemer to constantly check the connection to the Backend.

Cyberfly100, 2019-10-18:
I have not investigated further, but closing the progress popup on the frontend can cause the backend to get stuck, in the newest unreleased version, it seems. Could this be related to recent changes or was it just unlucky timing?

linzuzeng, 2019-10-18:
I am trying to reproduce it but haven't seen it happening yet.
Were you using realtime recipe?

linzuzeng, 2019-10-18:
Also if the backend is running, the frontend doesn't show a button for close. It will show the close button only when the backend finishes the processing.

Cyberfly100, 2019-10-18:
true.. I have no idea how it was possible. It was with the new realtime recipe, but I cannot reproduce it.

linzuzeng, 2019-12-09:
close for now. Reopen when it happens again

closed 2019-12-09 by linzuzeng