timetag/ETA #95: Instrument designer should allow scientific notation for Tools

Cyberfly100, 2019-09-17:
The intrument designer should be upgraded to understand scientific notation
e.g. HISTOGRAM('h1', 1e8, 1E5), where e and E are both *10^
This way variables on the front panel could also use this notation, since python in the code panel already understands this notation. It would prevent mistakes where e.g. one zero is missing.

linzuzeng, 2019-09-19:
I am trying to support the scientific notation by using int() to convert them into strings. 
I think we also need to sate in the documentation that the number has to be a integer instead of a float, although scientific notation supports both.

linzuzeng, 2019-09-19:
supported, have a try :)

Cyberfly100, 2019-10-14:
With the unreleased backend I get:
```
Compiling group HHT2...
ERROR:__main__:tuple index out of range
Traceback (most recent call last):
  File "C:\opt\Miniconda3\envs\etatest\Lib\ETA-master\etabackend\eta_runtime.py", line 108, in compile_eta
    etaobj, info_emitter)
  File "C:\opt\Miniconda3\envs\etatest\Lib\ETA-master\etabackend\etalang\eta_codegen.py", line 125, in compile_eta
    etavm.exec_eta(each)
  File "C:\opt\Miniconda3\envs\etatest\Lib\ETA-master\etabackend\etalang\eta_vm.py", line 45, in exec_eta
    command(*construct)
  File "C:\opt\Miniconda3\envs\etatest\Lib\ETA-master\etabackend\etalang\eta_exp.py", line 147, in HISTOGRAM
    each_converted = (int(each[0]),int(each[1]),each[2]) # supporting scientific notation
IndexError: tuple index out of range
```

Cyberfly100, 2019-10-15:
alright, the above bug is fixed and, for example, 1e4 instead of 10000 works.
But if I write 16 as 1.6e1 I get a ValueError: invalid literal for int() with base 10: '1.6E1'.
I think this could be fixed by using int(float('1.6e1')) instead of just int('1.6e1').

Cyberfly100, 2019-10-16:
Oh yeah, it's working and it's robust as well... works with negative numbers (does not crash) and decimal numbers, like 1.655e1.
Closed.