timetag/ETA #99: Correlate timetag streams from two different correlators

linzuzeng, 2019-12-09:
> I guess we should allow a file to be linked to an aquisition device. then file1 could be linked to aquisition device1 with channels 0-5 and file2 could be linked to aquisition device2 with channels 6-10.

The channel number assignment for two time tagger in ETA is not yet written, but we should definitely support it

Cyberfly100, 2019-12-09:
Zuzeng: In the ETA.run(), I would suggest using a dict like, {"devicename":"pathtofile"} so that it doesn't get too confusing about the correspondence of the files and the acquisition devices.
good idea, I think. it will not be immediately clear, which device belongs to which file on the home screen, though...

linzuzeng, 2019-12-10:
True. Shall we add more parameters to the Acqusition Device Panel on the home screen, like path to the file and the what kind of Time Tagger is being used. 

But then it would be confusing for batch mode and realtime mode...

linzuzeng, 2019-12-10:
We should somehow make it clear that the Acquisiton Device is something like a placeholder that should be filled with either a timetag file or a memory buffer (for true realtime mode without writing to disk).

Cyberfly100, 2019-12-10:
So far we have linked the aquisition device and file by just assuming there is only one and using ETA.run(filename). In hindsight, that was also not very transparent to the user.
Maybe we can get away with it again until we come up with a better way?
I like the way of linking them with a dict, but for it to be clear on the home screen, I think we would need to link them there and give only one of them to ETA.run().
Not sure what is better...

Chris645, 2019-12-10:
Maybe it'll get transparent as soon as another **Virtual Instrument** is added. It'll describe that the two files are combined into a new variable (a dict) called **cutfile_combined**. **cutfile_combined** is then transferred to **ETA.run()**. I'm not sure if that makes sense ~

linzuzeng, 2019-12-12:
> So far we have linked the aquisition device and file by just assuming there is only one and using ETA.run(filename). In hindsight, that was also not very transparent to the user.
> Maybe we can get [...]

How about this: we have only one acquisition device that serve as the "source" of timetags. All VIs should read from that source and emits on virtual channels. Some of the virtual channels can be set as "sink" and events on them can be written to disk for further processing.

In the case that users need to merge two timetags, they can add a combined source from two cut descriptors ,as suggested by

> Maybe it'll get transparent as soon as another **Virtual Instrument** is added. It'll describe that the two files are combined into a new variable (a dict) called **cutfile_combined**. **cutfile_combined** is then transferred to **ETA.run()**. I'm not sure if that makes sense ~

And since there is only one source, we can remove the "add acquisition device" button from the main GUI, and instead, put a new table for all the configurations about the source and the sink.

linzuzeng, 2019-12-12:
I wonder if we should change the "acquisition device" to a better name like "Timetag Source", "Event Source" or "Source & Sink", because we need to use  term "acquisition device" to refer to the individual timetag files that is later merged into one single "source". 

Also, we can add a configuration button for the "source" that allows users to configure the acquisition devices for that source.

Users many still add multiple sources into one recipe, but he need to specify in ETA.run, like `eta.run(name_of_source, ...)`

linzuzeng, 2019-12-12:
We should also add a tool to preview the combined source, since it might become complicated in the future, see #64

linzuzeng, 2019-12-12:
Now I wonder how would the simple_cut() and incremental_cut() should work with the combined source.

Assuing we use simple_cut, which **cuts the file in equal length in size**, we usually get **inequal length in time** because of the countrate fluctuation.
I drew a sketch to illustrate the problem here. Red is a timetag file, purple is another timetag file,  green line splits the simple_cuts.
[image]

In a non-parallel analysis mode where the context is mantained across each cut, this combined source will cause errors in most analysis, since the second cut has events that is older than the current time. It breaks the assumption that timetag can only grow in a monotonic order, which is the basic assumetion of almost all existing timetag analysis.

This problem will also prevent us from doing realtime on combined devices. 😭

linzuzeng, 2019-12-12:
To address this problem, I suggest to introduce Time-based cutting. Unlike simple_cut(), it happens after file combination. 

 See #100

linzuzeng, 2019-12-14:
I made a todo list in additon to the [milestone ](https://github.com/timetag/ETA/milestone/1)

- [ ] 1. RFILE tool(#104)
- [ ] 2. Introduce RFILE/VFILE ID assignment (replaces #64)
- [ ] 3. The sorting tree supports RFILE/VFILE ID in addition to channel number
- [ ] 3. Code generation for individual RFILE, replaces the hard coded C program
- [ ] 4. Add stop option (#58)
- [ ] 5. Make SYNC work again (#103)
- [ ] 6. Add WFILE
- [ ] 7. All GUI changes (#64 , #101)

gyger, 2020-02-17:
Whats missing for this issue to be used?

linzuzeng, 2020-02-17:
Actually we are quite prepared for supporting this feature. The only missing parts are
1. some code on Python side for parsing the new RFILE tool, and
2. automatically feeding multiple Clips from a dictionary of generators.

I have implemented the first part, but unfortunately, it was left on an offline computer in another city. 😢 I will try to merge it in and implement the second part as soon as I get back.

Also, I would suggest to release ETA 0.6.6 first, as it has many API changes already, and save this for 0.6.7 or 0.7.0

linzuzeng, 2020-02-23:
The first part is now in the multifile branch. All recipes haven't been updated, though. Only the Correlation recipe is updated.

linzuzeng, 2020-02-24:
#125 Merged to master

linzuzeng, 2020-02-24:
> Actually we are quite prepared for supporting this feature. The only missing parts are
>
>     1. some code on Python side for parsing the new RFILE tool, and
>
>     2. automatically feeding multiple [...]

linzuzeng, 2020-02-24:
> Actually we are quite prepared for supporting this feature. The only missing parts are
>
>     1. some code on Python side for parsing the new RFILE tool, and
>
>     2. automatically feeding multiple [...]

Both of them have been implemented.

Correlation of zero file is tested to be working, see the simulation recipe in the current master. :)
Haven't tested correlation of two files though, but I think it should work

@Chris645 sorry for letting you wait for so long

linzuzeng, 2020-02-24:
I made a sample recipe to correlate from two correlators. 

I don't have two files on my hand, so I made a recipe that has two RFILE tool to map the two channels from a same file to [0,1] and [2,3]. And then I use two dealy lines to further map [0,3] to [5,4].  The correlation results are exactly the same as running on one file with only chn [0,1].

[Correlation on multiple source.zip](https://github.com/timetag/ETA/files/4244154/Correlation.on.multiple.source.zip)

You can try setting a different path to correlate two files to correlate them

linzuzeng, 2020-03-10:
Changes is merged into master. :)

closed 2020-03-10 by linzuzeng