timetag/ETA #103: How would find last sync work in combined source?

linzuzeng, 2019-12-13:
Maybe it should take the SYNC from the first timetag file as reference.

linzuzeng, 2020-01-25:
We should allow users to set his own SYNC rate.

Cyberfly100, 2020-02-23:
aligning the syncs would be a great feature anyways. One file needs to be the master, other files internal time needs to be streched to match the master-sync. This becomes useful when the hardware has no external clock input to do it in internally (qutag and swabian have external clock input).

linzuzeng, 2020-02-24:
The current implementation in master is that if multiple T3 mode file are used,  will both starts at the 0. If the SYNC rate is the same , they are synced without extra efforts. If the SYNC are different, the find Last-sync will find SYNC from the first defined rfile (master).

If T3 and T2 are mixed, it should work similarly without extra configurations. Things get a little bit complicated if T2 doesn't start at 0 while T3 does. In order to mix them properly, you will need to manually modify the Clip object and add a negative value to global time shift (global means apply to all chns within the same source)

We need to put this in the documentation.

linzuzeng, 2020-02-24:
BTW, The new Clip system #113 also allow users to implementation their leap year policy #55 #59  easily with a customized leap year generator that modifies the global time shift per cut.

linzuzeng, 2020-02-25:
tested working. reopen if we find other issues.

closed 2020-02-25 by linzuzeng