timetag/ETA #114: Better verbose option

linzuzeng, 2020-01-29:
ETA can be called as a library. The caller might have its preferred way to display messages instead of eta.send() which works in the websocket environment.
The new verbose option will allow users to pass a reference to a Python function when calling it, and ETA will call this function with a string whenever it wants to display something. 

Verbose option is now default to `verbose=print`. Users need to use `eta.simple_cut(...,verbose=eta.send)` to get the default behavior of previous versions of ETA.

gyger, 2020-01-31:
I think the way to go would be to use python logging.

linzuzeng, 2020-02-01:
Good idea. But will it work with eta.send() that sends the message to the frontend using Websocket?

linzuzeng, 2020-02-01:
I think all ``verbose("...","err")`` message should be changed to ``raise Exception`` to allow users who use ETA as a library to handle the Exception. Other types of message should be handled by the ``message()`` function, which decides if the message should be send via ``eta.send``, or should go into the log. Then we can remove all ``verbose`` parameter from every API.

linzuzeng, 2020-02-01:
For users who use ETA as a standalone program. We can use a ``try ... catch``  in the backend.py  to stream the Exceptions to the frontend, like what we did before.

gyger, 2020-02-02:
I agree with using Exceptions. To contact the frontend I would use the logging facility aswell with a sublogger called e.g. etabackend.user? Then we register a logging handler to only this sublogger that forwards the messages to the frontend.

linzuzeng, 2020-02-02:
That would be great!

gyger, 2020-02-02:
I am working on rewriting that part.

linzuzeng, 2020-02-05:
I tried the version 8b4062c, and the logging and the new callback works great! :)

linzuzeng, 2020-02-20:
I was testing the realtime recipe, and found the logging too verbose. Can I set the loglevel or log filters to show only critical information on the backend or frontend?

gyger, 2020-02-20:
You can, import logging into the realtime recipe and then add:

```
logger = logging.getLogger('etabackend.frontend')
logger.info('No further logoutput for the realtime recipe.')
logger.setLevel(logging.WARNING)
plt.show()
logger.setLevel(logging.INFO)
```

linzuzeng, 2020-02-20:
Added to the manual
https://eta.readthedocs.io/en/latest/python.html#logging-getlogger-etabackend-frontend

Reopen if we find new problems.

closed 2020-02-20 by linzuzeng