timetag/ETA #118: attempted relative import with no known parent package

linzuzeng, 2020-01-31:
216ecdb2c38010ec495085da84ebd2fae17080bd seems to break ETA when launching directly from the debugging folder using ``python eta.py`` or ``python backend.py``

I know ``python -m`` would solve this, but it always call the ETA in the Python site-packages folder instead of the current git folder.

@gyger  any idea about how to solve this?

linzuzeng, 2020-01-31:
I tired for a while and couldn't find a good solution without making another hack.

I have two folder of different versions of ETA for comparison, but I don't want to change PYTHONPATH or reinstall ETA every time I switch. Maybe we should still allow running individual .py files directly.

gyger, 2020-01-31:
I would suggest installing ETA using pip -e . in the setup folder.

This development mode installs the package while still using your files
from the folder.

I'll check how to make the direct run possible aswell.

gyger, 2020-01-31:
This is such a pain, allowing scripts and package at the same time.
Relative imports are also not the ideal way of writing python code.

gyger, 2020-01-31:
As a reference:
https://stackoverflow.com/questions/16981921/relative-imports-in-python-3

linzuzeng, 2020-01-31:
Great! 
Please also remove other ways to launch ETA mentioned in #119

linzuzeng, 2020-01-31:
BTW, how about other scripts? 
I mainly use VS code, which runs the individual script instead of the package by default. 

I guess I need to find how to make a ``launch.json`` so that VS Code runs ETA properly.

gyger, 2020-01-31:
Fixed 907420f

gyger, 2020-01-31:
> BTW, how about other scripts?
I can not follow? What scripts?

gyger, 2020-01-31:
> I mainly use VS code, which runs the individual script instead of the package by default.
> 
> I guess I need to find how to make a `launch.json` so that VS Code runs ETA properly.

You can find more information here, how to debug a module instead of a file. `launch.json` is the way to go.
https://code.visualstudio.com/docs/python/debugging

linzuzeng, 2020-01-31:
> 
> 
> > BTW, how about other scripts?
> > I can not follow? What scripts?

Never mind. I thought you will need many copies of this file to run each `.py` file when debugging. But then I realized that it's just my bad habit to put debug code in the ``if __name__= "__main__": ``part.

> You can find more information here, how to debug a module instead of a file. `launch.json` is the way to go.
> https://code.visualstudio.com/docs/python/debugging

Thank you! I have already added it to my VS Code. :)

linzuzeng, 2020-01-31:
I just tried ``pip install -e`` and it worked. But now I found I can not uninstall the development package. It keeps saying "WARNING: Skipping etabackend as it is not installed.", but I can still call python -m etabackend 
I tried `python setup.py develop --uninstall` and also manually removed the egg-link, but the link seems to be written somewhere else.

@gyger Have encountered this before?

linzuzeng, 2020-01-31:
Removed the ``easy-install.pth`` and solved the problem. Seems to be a issue of pip. I will try to upgrade it.

commit 907420fa (2020-01-31): packaging: Full qualified imports plus eta_script.py, a script that allows running ETA from the directory.

We want to use ETA in two ways. Installed as a module, used for normal user installation but also in future automatic tests. For
local development it is also useful to launch ETA from a folder (e.g. comparing two versions). For this we added the eta_script.py file, that messes with your sys.path.

Fixes #118.
closed 2020-01-31 by linzuzeng