timetag/ETA #120: Separate ETA and Backend to make ETA an independent callable library

gyger, 2020-02-03:
It is almost at the point that the backend server and the ETA library is quite separated.

@linzuzeng What do you think about having a Backend Class that contains the ETA Class as a member?

linzuzeng, 2020-02-04:
The BACKEND inherts ETA, simply because of the annoying `eta.send()`. Thanks to your efforts in 04b5e7a, I think we no longer need that, and having ETA as a member is clearer. :)

linzuzeng, 2020-02-04:
Also I think maybe we should rename ETA Class to ETA Engine, ETA kernel, ETA core or ETA executor, or at least we should call it other than simply ETA. 
ETA should be the name of the comple, standalone software, while the other name only refers to the library that actually does the number crunching.

gyger, 2020-02-04:
For now I leave all the naming as it is and separate them into two. In a second step we can think about clearer naming. (Although I like the idea to just import an eta class.)

About the naming, is there a reason some of the Classes are All-caps?

gyger, 2020-02-04:
Commit #8b4062c makes a big step forward. I do not fully close this yet, as some problems could come our way.

linzuzeng, 2020-02-05:
> For now I leave all the naming as it is and separate them into two. In a second step we can think about clearer naming. (Although I like the idea to just import an eta class.)

We can also call the library itself eta, instead of giving it a new name like ETA engine.

> About the naming, is there a reason some of the Classes are All-caps?
Those in eta_exp are Tools, they are intended to be all-caps, so that CLOCK() and HISTOGRAM () look different from Actions, like c1.start() or emit ().
Other classes which have a name in all-caps are just arbitrarily choices, which can be rename without any problem.

linzuzeng, 2020-02-19:
I was trying to get `eta.send` working, but it seems that now eta in the recipe points to the kernel only. https://github.com/timetag/ETA/blob/master/etabackend/backend.py#L206

Maybe we need to add a `"backend":self` so that backend.send()  and other APIs will work like before

gyger, 2020-02-20:
I fixed injecting send into the kernel in 556a91ab04997c2ad89602d18a0753478899d9bc.

linzuzeng, 2020-02-22:
I have just updated the [documentation](https://eta.readthedocs.io/en/latest/advanced.html)

linzuzeng, 2020-02-23:
reopen when we find new problems

closed 2020-02-23 by linzuzeng