timetag/ETA #121: Script panel only displays two rows of code in Firefox

afsa, 2020-02-14:
**Steps to reproduce:**

1. Open the ETA github site [https://timetag.github.io/ETA/gui/src/renderer/#](https://timetag.github.io/ETA/gui/src/renderer)

2. Click on code on one of the script panels

**Expected result:**
The script panel should fill the full screen and display as many rows as possible.

**Actual result:**
[image]

**Browsers affected:**
- [ ] Chrome
- [x] Firefox

linzuzeng, 2020-02-17:
Sorry for the late reply.

I can reproduce it. I also found that if you resize the browser window, it will become normal again.

linzuzeng, 2020-02-20:
Fixed. Now it displays correctly on both Chrome and Firefox. 
If you are using the ETA Web GUI, try clearing the browser cache to get the latest version, as Github CDN might ask your browser to cache the assets files.

commit 52a73a23 (2020-02-20): gui: fix #121
closed 2020-02-20 by linzuzeng