timetag/ETA #123: Generators for Realtime recipe

linzuzeng, 2020-02-17:
ETA 0.6.6 has been in the draft for two weeks. Maybe we could release it at some point.

I tested the recipes except for the realtime recipe, and they works well. But I guess the realtime recipe would require some updates, as the `wait_for_data` has been integrated into [the new API](https://eta.readthedocs.io/en/latest/python.html#eta-clip-from-file-filename-modify-clip-none-read-events-0-format-1-wait-timeout-0-verbose-print). `ctx` is changed to `modify_clip`

@gyger @Cyberfly100 , could you help update and test the realtime recipes when you have time?

Cyberfly100, 2020-02-17:
Sure can do!

linzuzeng, 2020-02-20:
It seems that the realtime recipe will now crash the backend.

> This crash comes because it seems the TK backend is not closed when closing the figure and thereby ending the execution. The next time the script is run tk tries to close its previous loop from the wrong thread.

I made a standalone script for the realtime recipe, and it doesn't crash. Looks like it is related to the backend.py rather than the recipe itself

[ETAtestscript-new.zip](https://github.com/timetag/ETA/files/4229060/ETAtestscript-new.zip)

It seems that  websocket in backend.py will execute the script panel in the main thread for the first time, and in a different thread when the second client is connected...

Can we close the tk loop when the window is closed?

linzuzeng, 2020-02-20:
I also wonder why this doesn't happen to [previous versions](https://github.com/timetag/ETA/releases/tag/v0.6.5). 

We haven't update our websocket or backend implementation, and disentangling backend and eta libaray shouldn't cause this problem.

linzuzeng, 2020-02-20:
I found something strange in the log. Here is how I operated and the corresponding log.

1. Start front end.
> INFO:etabackend.backend:New client ('127.0.0.1', 52944) connected to port 5678.

2. Click run, the recipe script panel starts. The matplotlib shows and blocks the websocket.
> INFO:etabackend.frontend:Executing code in Script Panel in group main...
 ...
INFO:etabackend.frontend:Don't forget to save the recipe and share it!

3. Here the exec() of the recipe returns. So the websocket is processing incoming packets again. Close the frontend logging panel, click run again.

```
INFO:etabackend.frontend:ETA.clip_from_file: The file '\\\\HH400_T3_80MHz_HBT.ptu' section [4392,16392) is loaded into the Clip.    
INFO:etabackend.frontend:Compiling instrument group main.
INFO:etabackend.frontend:ETA.RUN: Starting new analysis using Instrument group main.
INFO:etabackend.clip:Initializing context.
INFO:etabackend.clip:Executing analysis program...
Exception ignored in: <function Image.__del__ at 0x000002053556C158>
Traceback (most recent call last):
  File "\\\\\Python37\lib\tkinter\__init__.py", line 3507, in __del__
    self.tk.call('image', 'delete', self.name)
RuntimeError: main thread is not in main loop
Tcl_AsyncDelete: async handler deleted by the wrong thread
```
However, in step 2, if I close the matplotlib right after it shows up , I can sometimes run script panel for the second time without seeing the error in step3.

linzuzeng, 2020-02-20:
I also noticed that if I wait for several minutes after closing the matplotlib window in step3, it won't show error. Very strange...

linzuzeng, 2020-02-20:
I made a minimum recipe to test if matplotlib works in the websocket. 
 
[unnamed.zip](https://github.com/timetag/ETA/files/4229596/unnamed.zip)

It seems that I can not reproduce this issue with this minimum recipe, suggesting the websocket might not be the cause of the problem.

linzuzeng, 2020-02-20:
I also checked all Python threads in the python.exe process using a debugging tool. websocket is not creating a second Python thread for the second matplotlib window. 

There is only MainThread and a bunch of ThreadPoolExecutor-0_*. ThreadPoolExecutor is created by eta.run() for parallel analysis.

linzuzeng, 2020-02-20:
Now I wonder if it is related to Python GC ...

https://bugs.python.org/issue18141
https://mail.python.org/pipermail/tkinter-discuss/2019-December/004148.html

linzuzeng, 2020-02-20:
I found another possible reason: library leak. It seems that the library imported inside the code of `exec()`, will leak to the outside or the next `exec()` call, and it can not be un-imported. 

I made a recipe to demonstrate this bug. `realtime-agg` is modified from the working `realtime` script, and it uses matplotlib Agg backend, which causes problems. 

[Realtime-libraryleak.zip](https://github.com/timetag/ETA/files/4230345/Realtime-libraryleak.zip)

If you run `realtime` first, the matplotlib window will show up, and then it will show a error on backend when you run `realtime-agg`, as expected.

If you run `realtime-agg` first, matplotlib will be configured to use Agg backend, which leads to the error as expected. After the `exec()` finished, all the global and local variables are destroyed, also as expected. Then you can click `realtime` to import matplotlib again, but it seems to be the previous matplotlib has come back to live. **Libraries are not destroyed with local variables!**

linzuzeng, 2020-02-20:
> 
> 
> Now I wonder if it is related to Python GC ...
> 
> https://bugs.python.org/issue18141
> https://mail.python.org/pipermail/tkinter-discuss/2019-December/004148.html

This error won't occur on Python 3.8.

We can add a Python version check at the end of the script to ask Python 3.7 users to reboot the back-end after each run, as a temporary solution.

 Maybe we should also say "drop Python 3.7 support" in the release information, telling people to upgrade their Python.

I am still looking for a workaround by clearing up the all the tk objects before the next exec() is called. But I am not super motivated since Python 3.8 will solve it in a proper way😅

linzuzeng, 2020-02-20:
I also tired wrapping everything in a `exec()`, so that it works in a environment like in the backend. Still couldn't reproduce the problem.
[ETAtestscript-exec.zip](https://github.com/timetag/ETA/files/4231699/ETAtestscript-exec.zip)

linzuzeng, 2020-02-20:
I finally reproduced the problem without the websocket environment! `backend.py` is to blame.

[reproduce.zip](https://github.com/timetag/ETA/files/4231743/reproduce.zip)

linzuzeng, 2020-02-20:
I found a minimal script to reliably reproduce the issue on my computer with both python 3.7.1 and python 3.8.1

[two-ways to reproduce.zip](https://github.com/timetag/ETA/files/4231936/two-ways.to.reproduce.zip)

linzuzeng, 2020-02-20:
I found a minimal script to reliably reproduce the issue on my computer with both python 3.7.1 and python 3.8.1.

The bug disappeard when GC is off. 

[gcbug.zip](https://github.com/timetag/ETA/files/4232057/gcbug.zip)

linzuzeng, 2020-02-20:
https://stackoverflow.com/questions/44781806/tkinter-objects-being-garbage-collected-from-the-wrong-thread

Looks like an ancient issue remain unsolved.

We can call gc.collect() after plt.show() to force collect stuff. I guess the recent changes of ETA somehow makes GC more confusing.

gyger, 2020-02-20:
Thanks a lot for troubleshooting this bug.

gyger, 2020-02-20:
I am wondering if at the end of the exec  we should just run the garbage collector in the backend.

linzuzeng, 2020-02-21:
> I am wondering if at the end of the exec we should just run the garbage collector in the backend.
I will have a try today. If it works well we can close this issue for now.

linzuzeng, 2020-02-21:
No it doesn't work.

linzuzeng, 2020-02-21:
I guess our structure of referencing objects and the exec() somehow make GC confusing. 

If the GC operation happens when a thread in background is executing(usually an analysis thread created by eta.run), tkinter will complain that it is not being collected from main thread.

We were using multiprocessing in previous versions, so this problem hides perfectly.

 In order to support streaming analysis, and small steps of cutting in incremental_cut(), we switch to threading, so that we don't need to launch a new python process send pickled objects between processes, reducing overhead of each eta.run().

We can maybe run matplotlib in a separated thread or process, not sure if that is easy...

linzuzeng, 2020-02-21:
An easy fix of this problem is to avoid creating threads, when `return_results=True` is set in `eta.run()`

If returns result is true, eta.run() will create a thread in the thread pool and then block on this thread, which is completely meaning less.

I will try to change this now, let's see if it solves the problem.

But still, this will not solve the problem if both multithreading and matplotlib are used in the same recipe.

linzuzeng, 2020-02-21:
Problem solved (hided) in 92ad68f985eecaaeee3b166f62d45b5f9a476da7.
Hurray!!

closed 2020-02-23 by linzuzeng