timetag/ETA #124: Recipe assembler

linzuzeng, 2020-02-21:
We should make a recipe assembler that assembles the python script into display panels in a recipe .

The current .eta format is not very git friendly. The JSON formatted string uses /n /r which is not recognized by git...

We could have it as a development tool and put both the .eta file and the source on GitHub, so that ETA Web GUI will continue to work like before. Just remember to run this tool once after making any changes in the recipes.

linzuzeng, 2020-02-21:
The standalone .py source file will also be good for #108
Users can Create a new Realtime Script Panel as a starting point for his own recipe.

gyger, 2020-02-22:
I think we should have a python script that can program the recipe. If it is ok I will brainstorm a little bit an suggest something on beginning of next week.

linzuzeng, 2020-02-22:
@gyger we have some already actually. But it haven't got the ability to modify the entire recipe yet. Now it can only modify the `table` part of the recpe.

https://eta.readthedocs.io/en/latest/python.html#modify-recipies-programatically

Each ETA recipe is a Json file consist of only one `table` for main screen, and multiple .py and . etainst files at its top level. Each file has an uniquely assigned ID, and it is referenced by one row in the table.

commit 9da86efe (2020-02-25): fix #124
closed 2020-02-25 by linzuzeng