timetag/ETA #133: Use bokeh for realtime display, drop tkinter

linzuzeng, 2020-03-11:


gyger, 2020-03-13:
I worked a little bit on it and it looks already decent, although can still be improved. Find a first draft attached. Needs current HEAD
[BokehCorrelation.txt](https://github.com/timetag/ETA/files/4331868/BokehCorrelation.txt)
The attached file is a .eta file.
.

linzuzeng, 2020-03-14:
Great! I have integrated it into the built-in correlation recipe.

gyger, 2020-03-14:
It is was only a draft, probably not yet ready to be in the built-in recipe. Did you change anything when moving to the built in recipe? I worked a little bit more on the code.

I also think we need to make our recipes saner. Perhaps have a choice when you press new what timetagger hardware you want to use.

I added one button to switch between alignment and accumulation mode.
[bokeh.txt](https://github.com/timetag/ETA/files/4332653/bokeh.txt)
Also cleaned the rest of the recipe a little bit more.

linzuzeng, 2020-03-15:
No I simply copy paste the code into the display panel. Doesn't require any changes on VI side since the histogram name (results) are the same.

Maybe you can directly put it into master as the new correlation recipe and rename the old correlation recipe to the development recipe:)

linzuzeng, 2020-04-21:
Do we have any features left on the tkinter implementation? If not we can close this issue. :)

Cyberfly100, 2020-04-22:
I think it is ok for the correlation recipe, now.  At least I cannot remember if I needed any more features. Bokeh can switch between accumulation and live and it can save. I added the header feature to the live save in correlation. We should also add it for live-livetime, and osci-mode.

Cyberfly100, 2020-04-22:
I added the header for lifetime and osci as well... I do not have a file to test osci, though. @gyger could you test if it works? Also it has a bin_resolution in the save txt code, I have not added this to the realtime version since it is not used there. Maybe this recipe could use some love anyways?

Cyberfly100, 2020-04-23:
aha, I remember the issue now! we are missing a lin/log toggle. The problem was, that in bokeh, all the interactivity is tied to the axes and they cannot just be changed on the fly. The suggested solution was to draw two plots (one lin, one log) and use a button to hide/unhide them or to plot them in two tabs.

linzuzeng, 2020-06-12:
We may also remove the requirements of dash (plotly) and lmfit, as we are using bokeh by default

linzuzeng, 2020-06-13:
Shall we close this and release 0.7.4? :)

gyger, 2020-06-13:
Fine with me. Lets then in a next step fix the recipes. I would suggest we make one demo recipe that contains all the information if one wants to use plotly or matplotlib tkinter and the other recipes we streamline to have only bokeh and matplotlib png export left.

closed 2020-06-14 by linzuzeng