timetag/ETA #134: clip generator with mutlithreading progress visualization support

linzuzeng, 2020-03-12:
Currently we show percentage of file reading in clip generators, but it is super noisy and doesn't make much sense when used together with muli-threading.

I would suggest to have a block diagram to visualize the analysis progress. I have an example from AiraNg
[image]

We just need to send the starting recordid and the ending record id of a file to the Web GUI, and ask Web GUI to put green on to the corresponding block, indicating that block of file is read already.

linzuzeng, 2020-03-12:
We could have a special logger for progress visulization, or add some javascript to parse the progress in the GUI side.
https://github.com/timetag/ETA/blob/master/etabackend/clip.py#L162

gyger, 2021-03-18:
There is a blog entry that has a similar structure: https://towardsdatascience.com/stunning-tables-using-bokeh-and-svg-e612a6f1c541

linzuzeng, 2021-03-19:
COOL!
Can bokeh render to a given div or we use iframe with a url pointing to a bokeh dashboard?

Cyberfly100, 2021-03-19:
https://docs.bokeh.org/en/latest/docs/user_guide/embed.html

It looks like it can target a specific <div>

>

linzuzeng, 2021-03-20:
Great! 
Now I realize maybe it's better to render the progress on the display panel (which is already bokeh) instead of embedding it on the main GUI.
It will allow users to see the progress on file with the the growing histograms on the same screen during an analysis. This may be handy for diagnosing some problem. 
Also this may save one server for bokeh and probably prevents the headaches of running two bokeh dashboard simultaneously :)

linzuzeng, 2021-03-20:
The only drawback is that the progress is not enabled by default for every analysis and  the user need to manually write progress bar rendering in the display panel Python script. 
But I think it's fine as the users don't always need progress bar (progress bar for realtime is meaningless ; progress bars for batch process in parallel requires extra rendering and typesetting ) and we can put something in eta.tk to minimize users' efforts.

Cyberfly100, 2021-03-20:
Yes, this sounds like a good idea to me. We can include the toolkitted
progress bar in the standard recipes, where applicable.