timetag/ETA #135: multithread_clips

linzuzeng, 2020-03-13:
Parallel analysis in ETA 0.7.1 is super complicated to start, compared to simple_cut in previous version

The most difficult part is that users need to make many generators for each thread, and set the index for each of them, so that the file is splited into roughly equal sizes for each generators.

```
read_events=1024
index_range = int(math.ceil(os.path.getsize(str(file))/(file_byte_per_record*threads*read_events)))
generators = []
for threadid in range(0,threads):
    generator.append(eta.clips(file, read_events=read_events,keep_indexes=list(range(threadid*index_range,(threadid+1)*index_range))))
return generators 
```

Instead of this, I would suggest to make a multithread_clips that warps this and return a list of generators for each thread.

Shall we call it ``multithread_clips`` or something else? 

We have ``split_file`` that split the file into equal clips already. Unlike ``split_file`` that split the file into a certain number of equal size clips, this would ``multithread_clips`` would make many generators,  and each of them can clip the file into a configurable ``read_events`` size clip.

Cyberfly100, 2020-03-13:
multi_clip?

linzuzeng, 2020-03-13:
multi_clips_list()?

commit 0b143bd2 (2020-03-14): library: fix #135
closed 2020-03-14 by linzuzeng