timetag/ETA #137: backend not found popup opens older backend

Cyberfly100, 2020-03-19:
probably not an issue for people who only have a stable version of ETA installed, but since I have both a stable and experimental backend I see room for improvement here.

The popup opens the stable backend by default (0.6.5) which lives in my anaconda environment 'ETA'. It is, however, not compatible with the GUI that the popup is coming from (7.1 latest). For this I have a second anaconda environment called 'etatest'. The popup should either allow selection from environments or it should automatically compare version numbers (or both?).

linzuzeng, 2020-03-20:
The pop up always open using ``python -m etabackend`` , currently.
I think we shouldn't scan users' computer for all versions of ETA backend. 
We can instead provide a bat file that allows users to customize the command to be executed

linzuzeng, 2020-03-20:
fixed. Update the install.rst in the documentation for how to configure ``startbackend.bat`` for conda users.

closed 2020-03-20 by linzuzeng