timetag/ETA #138: frontend timeout while looking for backend is too fast

Cyberfly100, 2020-03-19:
when I open the frontend and backend with successive double clicks, the frontend has already given up looking for the backend before the backend is ready to receive a connection. This timeout should be a bit slower, I think.

linzuzeng, 2020-03-20:
It's the default time-out of websocket. Do you know how to configure it?

Cyberfly100, 2020-03-20:
I found these two when checking the index.html for CSS styles:
1)
[image]
2)
[image]
Could those be the timeouts we are looking for?

Cyberfly100, 2020-03-20:
https://stackoverflow.com/questions/29881957/websocket-connection-timeout

linzuzeng, 2020-04-01:
fixed by updating electron 2bf4f5855a4d7f2b92b0c4de7a0941cbdc7e5b6b

closed 2020-04-01 by linzuzeng