timetag/ETA #139: fix the lin/log issue in bokeh

Cyberfly100, 2020-03-19:
bokeh seems to have all kinds of functionality attached to the axes (panning, zooming?), so it does not allow to easily change between lin and log scale. This functionality has proven very valuable in the lab, though.
Options for a workaround include making two graphs and 
1) toggling visibility between them (suggested by the developers somewhere) or 
2) making tabs for switching between two graphs.
Not sure if the realtime performance or performance with lots of data points can be maintained with two plots. Maybe with a smart implementation...

gyger, 2020-06-08:
I think this is mostly done. Should just be deployed in all the recipes

Cyberfly100, 2020-06-08:
there is also this suggestion: https://github.com/bokeh/bokeh/issues/6575#issuecomment-312446284
which allows the use of a button and might perform better.
the tab solution is already implemented in the static displays. It still needs to be implemented for the realtime displays, though.

linzuzeng, 2020-06-09:
> I think this is mostly done. Should just be deployed in all the recipes

Can we create a class in etabackend for this kind of the display panel, so that we write less in each recipe?

Cyberfly100, 2020-06-09:
excellent suggestion. I'll think about which parameters we need. I think
for lifetime, correlation and simulation the requirements are the same. and
realtime panels also have very similar requirements.

gyger, 2020-06-11:
I would suggest we make a package called etabackend.tk (for toolkit) and there e.g. a module plot that encapsulates plotting. Then we load the toolkit into the recipe workspace.

gyger, 2020-06-11:
Here are some code snippets that can be used for the recipes.

    fig = column(plin)

    button_linlog = RadioButtonGroup(labels=["Linear", "Logarithmic"], active=0)
    button_linlog.on_click(lambda nv: bokeh_button_linlog_callback(fig, plin, plog, nv))

    def bokeh_button_linlog_callback(col, flin, flog, new_value):
         # See https://github.com/bokeh/bokeh/issues/6575#issuecomment-312446284
         if new_value == 1: # Log
             col.children[0] = flog
         elif new_value == 0: #Lin
             col.children[0] = flin

where flog and flin are the two figures that are replaced

Cyberfly100, 2020-06-12:
has been implemented into the static plots of correlation, lifetime and simulation.

closed 2020-06-12 by Cyberfly100