timetag/ETA #141: add hovertool to bokeh script panels

Cyberfly100, 2020-03-21:
to have feature parity with pyplot we should include the hovertool in at least the static bokeh plots.
countrate should already have it so we can just copy it from there.

Cyberfly100, 2020-03-31:
added to correlation. Unfortunately the hovertool does not work with the step-glyph, so we are back to curve

Cyberfly100, 2020-05-19:
I found it in correlation, simulation, countrate and lifetime static displays, but not in Osci nor any of the realtime displays

linzuzeng, 2020-07-07:
Is this fixed already, after the most recent recipe update in 0.7.5?

Cyberfly100, 2020-07-07:
not for the ones under "more" in the recipe selector. But with the toolkit this should now be taken care of kind of automatically once these recipes catch up.

closed 2020-07-07 by Cyberfly100