timetag/ETA #143: frontend version needs to be automatic

Cyberfly100, 2020-05-03:
the frontend version after the update (did not work automatic, though I got impatient after 5 min) is still 7.2.0 instead of 7.3.0. This should be automatic so that it is not necessary to remember that a manual change is necessary, there.

linzuzeng, 2020-05-04:
I seems that both pip install upgrade and the automatic upgrading upgrades from 0.7.2 to 0.7.3 correctly. 

[image]

Maybe anaconda or virtualenv is doing something strange here?

gyger, 2020-05-11:
I think you need to reload with CTRL + F5, i had similar problems. We cache the webpage in the browser, not sure if this is something we need to change.

Cyberfly100, 2020-05-19:
seemed to work well this time. I think we can close.

closed 2020-05-19 by Cyberfly100