timetag/ETA #144: Fix the automatic update of the backend

Cyberfly100, 2020-05-19:
currently, the automatic update always fails for my anaconda installation. I am fine with installing it myself, but I think ETA should learn when installed initially, that I am using anaconda and just keep the frontend up to date. It could maybe show a promt to remind me to update the backend myself.

linzuzeng, 2020-06-08:
The updating logic is hard-coded here 
https://github.com/timetag/ETA/blob/be63d452a43f77ac0561b46541b3338efac55dfc/gui/src/main/backend.js#L57

We can maybe use a .bat file instead, like
https://github.com/timetag/ETA/blob/be63d452a43f77ac0561b46541b3338efac55dfc/gui/src/main/backend.js#L105

linzuzeng, 2020-06-08:
@Cyberfly100 The new version is in draft, could you have a test?

linzuzeng, 2020-06-13:
Currently `startbackend.bat` uses pip to install all dependencies. It breaks anaconda installation easily if the required version matplotlib/numba is higher than the existing one installed by conda.

Can we somehow in the .bat file detect if anaconda is being used, and try install the `requirements.txt`  with codna first, ignoring the missing packages, and then try installing with pip to install the rest?

Cyberfly100, 2020-06-13:
In my anaconda install there is a separate environment for ETA (called ETA) with only pip installed packages. Maybe we could detect if an environment called ETA (could also be specifiable by the user during install) exists and create one if not.

linzuzeng, 2020-06-14:
Sounds complicated...

closed 2021-09-28 by linzuzeng