timetag/ETA #145: Numba Error

gyger, 2020-06-03:
Today I got this Numba Error, seems something is coming upon us with
Numba 0.50.0. Should be an easy fix I guess.

Import requested from: 'numba.cgutils', please update to use 'numba.core.cgutils' or pin to Numba version 0.48.0. This alias will not be present in Numba version 0.50.0.

Cyberfly100, 2020-06-07:
so we just replace all nb.cgutils with nb.core.cgutils?

linzuzeng, 2020-06-08:
We didn't use `nb.cgutils`. Maybe it's a problem from numba?

Cyberfly100, 2020-06-08:
[image]
I found it in the jit linker

linzuzeng, 2020-06-08:
Sorry, I forgot 

> so we just replace all nb.cgutils with nb.core.cgutils?

Will it work after replacing?

Cyberfly100, 2020-06-08:
haven't tried. was too afraid to break something

gyger, 2020-06-08:
From the error message it was just an alias. So I guess it should work. Not sure though.

linzuzeng, 2020-06-08:
Fixed this, but I get more like this. :(
 
https://github.com/timetag/ETA/runs/749882126#step:5:117

closed 2020-06-08 by linzuzeng

Cyberfly100, 2020-06-08:
maybe numba 0.49 is using numba.typing itself somewhere?