timetag/ETA #147: Bundle repetitive code in toolkit for ETA recipes.

gyger, 2020-06-11:
As discussed in #139 we should bundle repetitive code in the backend library.

I would suggest the following, and have started with a small implementation.

We add an extra subpackage called `tk` for toolkit and in there we add several modules for the different usecases. I started now with one called plot.py to add support functions for plotting. I was inspired by the plot utilities von [arlpy ](https://github.com/org-arl/arlpy/blob/master/arlpy/plot.py) and I like the idea. 

I started with two functions:

- beautify(figure)
Dose all the styling of the figure

- interactivity(figure)
Adds the extra tools and perhaps also save button and so.

Another module could be data.py to help with data handling. I though perhaps the merging of the histogram data, not sure yet though.

gyger, 2020-06-12:
See Commit e4f904f for a first step.

linzuzeng, 2020-06-13:
Great! 
The only thing is that `tk` makes people think of tkinter. Maybe `etaplt` or simply `plt` is a better name?

gyger, 2020-06-13:
I would like to bundle also other code (I thought of e.g. fitting g2) in the toolkit at one point, so I did not want to call the whole part plot.
I dont think it is a big problem, that it resembles tkinter, because I did not expose it in the main namespace, but it is anyway below etabackend.tk.

If you do not agree, i would suggest etk then for eta toolkit.

linzuzeng, 2020-06-13:
`etk` is a bit redundant. Let's use `tk` then :)

gyger, 2020-06-13:
Lets close this and add what we need later.

closed 2020-06-13 by gyger