timetag/ETA #150: Use node red as VI orchestration UI

linzuzeng, 2020-06-23:
Maybe we can try node red as VI orchestration UI. 
Each VI can be represented as a node in the graph, and the numbers of input channels and output channels  can be simply shown as connectors for each VI. 

https://nodered.org/
and here is an example 
https://www.pjrc.com/teensy/gui/index.html

gyger, 2020-06-28:
While I do like the flow, I am against putting Node into the package. It becomes too complicated and bloated.

linzuzeng, 2020-06-28:
No we don't need Node. We only need thier frontend library.

linzuzeng, 2020-06-28:
> The channel number in ETA was designed as an analogy of the channel number in a real timetagger device. Initially we thought it is nice to give real channels in lower numbers from 0, and leave non-exisiting higher numbers for virtual channels.
> 
> However, if we go the LabVIEW-way eventually, we don't need such channel number to connect different VIs together. 
> We can use diagram with wires instead, and in the state diagram of each VI we only need to refer to the connector name.

> The only drawback of the graphical wiring ui is that, it makes * ("catch all") looks like a mess. You will need to connect every channels from time tagged to every VIs, basically. With "catch all" you only need to type one symbol

gyger, 2020-06-28:
Ah ok, yeah in that sense splitting of the frontend of their code seems to be an interesting way to go. (I am still dreaming about lab orchestration in the the future, so their such a library would become very handy.

linzuzeng, 2022-05-06:
https://reactflow.dev/ Seems to be the new cool thing.

Maybe we could use their https://www.npmjs.com/package/react-flow-renderer as an alternative to our existing table-based main GUI.