timetag/ETA #151: Anaconda requirements file

linzuzeng, 2020-07-09:
I just found that one can create a 'pip:' section in environment.yml for conda.

Maybe we can add an environment.yml to ETA, for easier installation with `conda env create -f environment.yml`

Here is an example for the environment.yml

```
name: ETA
channels:
  - conda-forge
  - defaults
dependencies:
  - python=3.8.1=h5fd99cc_1
  - xxxxx
  - pip:
    - etabackend==0.7.xxxx
```

Cyberfly100, 2020-07-09:
yes, mine looks like this:

name: ETA
channels:
  - conda-forge
dependencies:
  - python=3
  - pip
  - pip:
    - etabackend

not entierly sure if it is necessary to ask for pip the exist before specifying what whould be installed via pip

linzuzeng, 2020-07-09:
Don't you install matplotlib and other libs from conda first? I thought installing matplotlib with etabackend  from pip would break the conda environment.

linzuzeng, 2020-07-09:
It works though. Close for now

linzuzeng, 2020-07-09:
The installing script is in 

https://github.com/timetag/ETA/blob/master/gui/upgradebackend.bat

and the environment checking logic is in

https://github.com/timetag/ETA/blob/master/gui/src/main/backend.js

Pay attention to `detached: false` and `detached: true, shell: true`

gyger, 2021-02-03:
I changed the conda file (bc593ae95e52fbad14650a055e2eba65d1bf0eef) to use mostly conda to install the packages, we only use pip for etabackend (that is the suggested way). [Best practices for mixing pip and conda.](https://www.anaconda.com/blog/using-pip-in-a-conda-environment).

linzuzeng, 2021-02-04:
Great job! Shall we use conda as default way to install ETA? Or we keep the pip-way and put the conda-way in the documentation.

related installation code in  https://github.com/timetag/ETA/blob/bc593ae95e52fbad14650a055e2eba65d1bf0eef/gui/upgradebackend.bat#L1

linzuzeng, 2023-02-16:
Shall we zip the conda venv folder (for Windows) so that people with poor or no Internet connection could run ETA with the bundle?

Also it helps sloving the mess created by having multiple offical python installations on the same computer.

gyger, 2023-02-17:
Yeah, this tool [https://conda.github.io/conda-pack/](conda-pack) could be a good way of distributing the environment around the installer.

Second, I recently discovered Jupyter-desktop, they seem to handle a lot around all this mess with electron and Python, perhaps we could somehow benefit from them.
https://github.com/jupyterlab/jupyterlab-desktop

linzuzeng, 2023-02-17:
Conda-Pack Seems to be a drop-in replacement at the moment.  

jupyterlab-desktop has Extension support, which could be used for making the GUI. but it's a bit complicated. The GUI is intended for users new to Python. If they already know Jupyter , they must know how to install ETA via pip themselves and they probably already has a jupyter installed.

I agree that using Jupyter as a replacement of the existing display panel could be a very interesting direction to go. We still ne to figure out how the jupyter GUI and ETA GUI could be combined.

linzuzeng, 2023-02-17:
Fixed via 1af53711f33d90698f12e0609bb75dbf11d8b0d3

closed 2023-02-17 by linzuzeng