timetag/ETA #155: Swabian has a "new" fileformat

gyger, 2022-05-02:
Swabian has a new fileformat, that starts somehow with strings and then the timetag data.
It can also be automatically split into multiple files, but I guess that can be treated in a second step.

I attached a sample file, the documentation is found [here](https://www.swabianinstruments.com/static/documentation/TimeTagger/api/Measurements.html?highlight=filewriter#filewriter), they like the file extension _ttbin_.
[test_2022-05-02_143549.1.zip](https://github.com/timetag/ETA/files/8605461/test_2022-05-02_143549.1.zip)
.

linzuzeng, 2022-05-06:
the doc seems to be unclear about how the file compressed. 

Instead of adding decompression of that format in our reader, maybe we can use their FileReader directly to obtain the uncompressed format (which is supported already) and construct the Clip object in the memory.

https://eta.readthedocs.io/en/latest/advanced.html?highlight=.buffer#read-a-header-less-file

https://eta.readthedocs.io/en/latest/advanced.html?highlight=.buffer#build-a-clip-without-any-file

gyger, 2022-07-01:
The new file seems to be encrypted. I discussed with them if there is a way to just access the timetags. Currently to run the library you need to connect a timetagger on the computer once to get the license.

gyger, 2023-02-17:
There is no solution but annoying Swabian Instruments not to deprecate the export in their old file format.

closed 2023-02-17 by gyger